cezanne/usbip-win #309: Autostart usbipd.exe on Windows

v-bulynkin, 2022-05-17:
Hello,

Could you please explain how to make usbipd.exe started on every boot automatically?

Everything is working fine if I kick it off by **.\usbipd.exe -d** (however, it doesn't detouch from active console), but I can't force it starting neither by making it a service (NSSM) nor by trying to start it through Task scheduler.

Moreover, what user can be used? That'd be great if it would be SYSTEM, NETWORK SERVICE or similar one.

Tbinn, 2022-10-25:
Same issue here. @v-bulynkin, maybe you have any progress in that?

Kogotoro, 2022-10-25:
> NETWORK SERVICE
probably no
https://serverfault.com/questions/116379/is-it-possible-for-a-scheduled-task-to-run-as-network-service