cezanne/usbip-win #310: "USBIP Test" certificate not found. Please install first!

bihailantian655, 2022-05-25:
i have already instal certificate  but still ger the  warning  

"USBIP Test" certificate not found. Please install first!

\![Uploading 3CZ)MUQ7KB@_~])99%ML[KE.png…]()

bihailantian655, 2022-05-25:
i have already instal certificate  but still ger the  warning  

"Trusted Root Certification Authority" in "Local Computer" (not current user) and
"Trusted Publishers" in "Local Computer" (not current user)

"USBIP Test" certificate not found. Please install first!
![Uploading 3CZ)MUQ7KB@_~])99%ML[KE.png…]()

bihailantian655, 2022-05-26:
[image]

bihailantian655, 2022-05-26:
![Uploading R{}N)0GVK{8]UA2HF[DH_WS.png…]()

cezanne, 2022-06-11:
@bihailantian655 : Weird.. Please try to install again with `-d` (debug) option like as `usbip.exe -d install`.

flejz, 2022-07-13:
Make sure you're running your powershell/cmd with admin privileges, otherwise it can not read from the certificate store.