cezanne/usbip-win #312: Blue Screen Problem

IasonInBactria, 2022-06-22:
Hi, recently after I successfully establish connection, I copied a large file from the client machine to my flash disk, the server side always encountered a blue screen crash with error code PFN_LIST_CORRUPT. I don not know how to debug this problem. My usbip version is 0.35. Could you teach me how to debug this driver issue or how to solve it? Thanks so much.

adrianhuuk, 2023-01-10:
I have a similar problem with my usb camera/microphone. My pc crashes if I use the mic for too long and the camera crashes it even quicker if I use it. It's almost as if there was some sort of data limit per session, which is filled up faster by video (or file transfer) than audio.

I'm not sure what the error code is as I use parsec and the error doesn't show on the virtual display.

592368293, 2023-03-17:
Can anyone solve this problem?
I am using the webcam on a Win10 client/server and the server crashes after a few minutes, I would like to know how to solve this error or how should I debug it, can anyone help?
thank you very!

adrianhuuk, 2023-03-17:
If I remember rightly, using this fixed the problem for me: github.com/vadimgrn/usbip-win2
I used version 0.9.3.2 (The latest at the time)
Hope that helps!