cezanne/usbip-win #315: using usb/ip with Amazon Kindle

tadabrowski, 2022-06-30:
Hi!

Has anyone tried using usb/ip with Amazon Kindle? 

Binding seems to be successful, but no drive shows.  Moreover I use a dedicated app of an online library - it synchronizes rented books with Kindle, but it says the Kindle is disconnected.

[image]
[image]

I use Windows 10 on the client side and Ubuntu Mate on the server side.

vadimgrn, 2022-07-01:
Try https://github.com/vadimgrn/usbip-win2, pls

tadabrowski, 2022-07-07:
I just did without any luck. I'm going to submit an issue there.