cezanne/usbip-win #318: "USBIP Test" certficiate not installed or properly located in certificate store

LeeHyunTae, 2022-07-25:
hello.
I was looking for usb over network-open source to be used from window to window system

I found your app by chance

Is the readme.md file still valid for installing a below version...?
PS D:\Git\usbip-win-0.3.6-dev> .\usbip.exe bind -b 1-156

I met a below error when installing usbip as ur guide(readme.md)
usbip: error: "USBIP Test" certficiate not installed or properly located in certificate store
could u check it....?

lemchen, 2022-07-26:
same issue.

lemchen, 2022-07-26:
https://docs.microsoft.com/en-us/windows/wsl/connect-usb

this may work, you should try.

Kogotoro, 2022-07-27:
admin console
```
CERTUTIL -f -user -p usbip -enterprise -importpfx root "usbip_test.pfx"
CERTUTIL -f -user -p usbip -enterprise -importpfx trustedpublisher "usbip_test.pfx"

```
or as README.md says:
```
Install USB/IP test certificate
Install driver/usbip_test.pfx (password: usbip)
Certificate should be installed into
"Trusted Root Certification Authority" in "Local Computer" (not current user) and
"Trusted Publishers" in "Local Computer" (not current user)
```

bejaranoo, 2022-07-27:
certutil -f -service -p usbip -importpfx Root "usbip_test.pfx" 
certutil -f -service -p usbip -importpfx TrustedPublisher "usbip_test.pfx"

Kogotoro, 2022-07-28:
without -enterprise its install it in current user certificate store
-enterprise = loacal computer certificate store

maybe ok only for user... depend on how it searched from code.

bejaranoo, 2022-07-28:
[image]
I've tested the command, and it install also on local machine you can check each store with Win + R, certlm.msc or certmgr.msc

> certutil -f -service -p usbip -importpfx Root "usbip_test.pfx" certutil -f -service -p usbip -importpfx TrustedPublisher "usbip_test.pfx"

These are the flags explained
[image]

Kogotoro, 2022-07-28:
ok.. my bad, thx for clarification =)

Kogotoro, 2022-07-28:
off top
mm...
http://certificate.fyicenter.com/685_Microsoft_CertUtil_Microsoft_certutil_-user_Certificate_St.html

no options
"HKEY_LOCAL_MACHINE\Software\Microsoft\SystemCertificates"

option: -service
"HKEY_LOCAL_MACHINE\Software\Microsoft\Cryptography\Services \ServiceName\SystemCertificates"

what wil be difference?

elvis-sun, 2023-03-27:
> my

@Kogotoro 

> ok.. my bad, thx for clarification =)

@Kogotoro     "usbip.exe bind" is ok now?

LAIhy, 2024-04-16:
CERTUTIL -f -user -p usbip -enterprise -importpfx root "usbip_test.pfx"
CERTUTIL -f -user -p usbip -enterprise -importpfx trustedpublisher "usbip_test.pfx"

I use these two CMD， problem solved.