cezanne/usbip-win #319: Signed driver?

josephdunne-eaton, 2022-07-26:
Is there any hope of getting a signed driver? My organization has policies set which disallow use of test signed drivers.

paulpv, 2022-09-21:
Furthermore, many games with ant-cheat enabled won't run if they detect Windows is in Test Mode.

Is there any reasonable workaround for PCs that trust this driver to not have to be run in Windows Test Mode?

XIAZY, 2022-09-25:
Ditto, I was trying to use the client in a Windows Docker container, therefore impossible to turn on test signing mode. Unfortunately I cannot proceed installing unsigned drivers.

ashleyw-gh, 2024-04-28:
The lack of a signed driver is sadly a no go for most people due to in built in windows security. Is there a possibility for the code to be signed by this organisation? It appears to be zero cost for selective open source projects.
https://signpath.org/
https://about.signpath.io/product/open-source
?