cezanne/usbip-win #320: Build for Windows 10+ Universal

MatthewTingum, 2022-08-01:
This fails to build out of the box with the following configuration:
- Target OS Version: Windows 10 or higher
- Target Platform: Universal

For this, we need:
- Spectre mitigation enabled
- Replace deprecated functions
- More?

Kogotoro, 2022-08-02:
vs2022-17.2.6
wdk-10.1.22621.1

in libdrv, usbip_stub, usbip_vhci, usbip_vhci_ude

Target OS Version: Windows 10 or higher
Target Platform: Universal
Spectre mitigation disabled

Replace deprecated 
ExAllocatePoolWithTag(PagedPool -> ExAllocatePool2(POOL_FLAG_PAGED
ExAllocatePoolWithTag(NonPagedPool-> ExAllocatePool2(POOL_FLAG_NON_PAGED

build every project separate, not build entire solution.

at least it builds... 
about working ... not sure cause 
```
// Old code
PVOID Allocation = ExAllocatePoolWithTag(PagedPool, 100, 'abcd');
RtlZeroMemory(Allocation, 100);

// New code
PVOID Allocation = ExAllocatePool2(POOL_FLAG_PAGED, 100, 'abcd');
```
and i dont saw RtlZeroMemory after ExAllocatePoolWithTag in old code mostly? only RtlCopyMemory sometimes.

Kogotoro, 2022-08-02:
ok... i just instlled last spectre libs, and it builds.

MatthewTingum, 2022-08-02:
The example is just showing that the new `ExAllocatePool2` function zeros the memory [for you](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exallocatepool2#remarks).
> Memory is zero initialized unless POOL_FLAG_UNINITIALIZED is specified.

The docs for `ExAllocatePool2` also say:
> If you are building a driver that targets versions of Windows prior to Windows 10, version 2004, use ExAllocatePoolZero, ExAllocatePoolUninitialized, ExAllocatePoolQuotaZero, or ExAllocatePoolQuotaUninitialized.

So maybe `ExAllocatePoolZero` is a better replacement for `ExAllocatePoolWithTag` because it has greater compatability.

MatthewTingum, 2022-08-03:
I'll put in a PR for this. My biggest concern is breaking existing workflows. The READAME states that this project is not production ready. There are tags and releases in place. Presumably, anyone using this in production is using a tagged or released build.

My plan is to:
- Replace ExAllocatePoolWithTag with ExallocatePoolZero
  - It's a drop in replacement
  - I assume this projects doesn't depend on pool allocations being non-zeroed
    - or executable for that matter
  
`ExAllocatrePoolZero` requires the following:
> To run on versions of Windows prior to Windows 10 version 2004, the driver must define POOL_ZERO_DOWN_LEVEL_SUPPORT and call [ExInitializeDriverRuntime](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exinitializedriverruntime) before calling this function.

This seems like a good idea, but I'm not familiar enough with the project to know if execution on the pool is necessary. If it is, we should fix that.

Kogotoro, 2022-08-03:
upd:
its build... but not work xD (win10, vs2022... and other staff described above)

on win10 it bind without errors , but device in manager shows with error =/
(same for Spectre mitigation enabled/disabled)

__
previosly i build it on win8.1, vs2019, wdk-10.0.18362.1
without changing 'Target OS Version','Target Platform'
only disable 'spectre', and it worked on win10...

... but my build have some experimental changes... so not certain.

Kogotoro, 2022-08-03:
and again upd...

tried rebuild on (win10, vs2022...)
usbip_stub, usbip_vhci ~ 'Target Platform: Desktop' (probably it also be ok if i will set 'Target Platform: Universal' , not checked)

!!! and dont replace ExAllocatePoolWithTag with ExAllocatePool2, 
instead i turn off 'treat warnings as errors'...

and it works now O_o?

Kogotoro, 2022-08-03:
> I'll put in a PR for this. My biggest concern is breaking existing workflows. The READAME states that this project is not production ready. There are tags and releases in place. Presumably, anyone [...]

just checked... it works.
my test VBox has win10.v18362 ... so ExAllocatePool2 not work because of this ^^'.

we also may need consider this...
https://www.osr.com/blog/2020/07/14/bug-in-new-function-exallocatepoolzero-results-in-security-vulnerability-and-crashes/
https://www.osr.com/blog/2021/01/07/mitigations-exallocatepoolzero-security-vulnerability/

MatthewTingum, 2022-08-05:
> !!! and i didnt replace ExAllocatePoolWithTag with ExAllocatePool2,
> instead i turn off 'treat warnings as errors'...

This isn't surprising.....
Deprecation of APIs generally involves deterrence of usage in the form of warnings followed by a complete drop of support.

> we also may need consider this...
[links]

The [official docs](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/nf-wdm-exallocatepoolzero) state thsi too. As you've stated, not everything zeros memory. I don't think we're missing much by not doing something that never happened.

Can we consider a test matrix for this project? There are no contribution guidelines and I don't  know when I should consider a PR ready for review.