cezanne/usbip-win #322: vhci driver weird code

Kogotoro, 2022-08-24:
/driver/vhci/vhci_devconf.c
function setup_intf(...)

intf->InterfaceHandle = TO_INTF_HANDLE(intf->InterfaceNumber, intf->AlternateSetting);

considering what 
#define TO_INTF_HANDLE(intf_num, altsetting)	((USBD_INTERFACE_HANDLE)((intf_num << 8) + altsetting))
and 
intf->InterfaceNumber, intf->AlternateSetting both UCHAR (ie 8bits and there 8 bits left shift on first of them?)

its eq to
intf->InterfaceHandle = (USBD_INTERFACE_HANDLE)(intf->AlternateSetting);

???

vadimgrn, 2022-08-24:
If you really want to improve usbip client, consider this [project](https://github.com/vadimgrn/usbip-win2/tree). This project is stagnating.

vadimgrn, 2022-08-24:
The author was working on UDE driver, so VHCI driver is double abandoned.

vadimgrn, 2022-08-25:
Regarding your question, read about integral promotions in C.

Kogotoro, 2022-08-26:
> Regarding your question, read about integral promotions in C. In expression intf_num << 8 a compiler promotes UCHAR to int and does shift.

thx

closed 2022-08-26 by Kogotoro