cezanne/usbip-win #323: Build fails

ShamimIslam, 2022-08-29:
Do I need to be on a Windows 10 machine to perform the build? Also, I have the 1903 SDK but the Signtool is not found. Thanks.

Kogotoro, 2022-08-30:
> Do I need to be on a Windows 10 machine to perform the build? 

Probably no. (i builded from win8.1 for win10 in vs2019)
(if entire solution not builds, then try build separate projects inside it one by one (initial project config probably have some small errors or missing targeting fields and sign settings) that can help locate errors in config)

>Also, I have the 1903 SDK but the Signtool is not found.

Signtools is optional feature while instaling sdk. ('Windows SDK Signing Tools for Desktop Apps')
Should be in:
c:\Program Files (x86)\Windows Kits\10\bin\10.0.18362.0\<platform>
platform x64,x86,arm...
try reinstall/fix and check what it selected.

paulpv, 2022-09-14:
I am running in to `Spectre-mitigated libraries are required for this project` errors, even after installing literally every single Spectre x64/x86 option there is in the VS2019 `Individual Components` tab. :/

paulpv, 2022-09-14:
Doh! I was installing the `Spectre` components for the `Visual Studio Build Tools 2019` install instead of the `Visual Studio Community 2019` install.
I added the `MSVC v142 - VS 2019 C++ x64/x86 Spectre-mitigated libs (Latest)` to my `Visual Studio Community 2019` install and it then compiled fine.
I removed the 30+GB of other installs I added to the wrong "2019" product and things still build fine.

ShamimIslam, 2022-10-11:
I found the binaries. Running from linux to linux, I get great throughput.
But running from Linux to Windows, the video halts after a while. Not sure
why.