cezanne/usbip-win #332: Will this work with Hyper-v?

Pomax, 2022-12-17:
Hyper-V notoriously hates anything USB related, will this work on windows guests under Hyper-V?

zeerayne, 2023-01-04:
Yes, it will. I'm currently testing USB client on Hyper-V guest an it works perfect. Only issue i can notice is quite long guest reboot/shutdown if it hasn't detached all usbip devices before.

Pomax, 2023-01-04:
Can you explain how you made that work? Because even if the proxying itself works, Hyper-V refuses to actually use any "this is not storage" USB device. It'll show up as shared, it'll show up in device manager, it will report working correctly, and then it also won't actually show up in a application usb device list. (e.g. no MIDI controller actually ends up being usable this way)

zeerayne, 2023-01-04:
Have to note, that I don't use usbip server on windows, it is on linux host, and I only use usbip client on Hyper-V guests.

Steps to reproduce my usbip config:

1. Build executables using guide in [README.md](https://github.com/cezanne/usbip-win/blob/master/README.md)
2. Copy `usbip_test.pfx` to Hyper-V guest (e.g. in C:\usbip)
3. Import cert to root store `certutil -f -p "usbip" -importpfx "Root" "C:\usbip\usbip_test.pfx"`
4. Import cert to trusted publisher store `certutil -f -p "usbip" -importpfx "TrustedPublisher" "C:\usbip\usbip_test.pfx"`
5. Turn on testing mode on guest (need to disable secure boot in Hyper-V VM settings to do this) `bcdedit.exe /set TESTSIGNING ON`
6. Reboot
7. Copy client files to Hyper-V guest: `usbip_vhci_ude.cat`, `usbip_vhci_ude.inf`, `usbip_vhci_ude.sys`, `usbip.exe`, `attacher.exe`
8. Install UDE driver `C:\usbip\usbip.exe install -u`
9. Attach usbip device `C:\usbip\usbip.exe attach -r <linux_usbip_host> -b "<some_usbip_device>"`

Pomax, 2023-01-05:
Yep, did all of that, and while the devices claim to bind, the Hyper-V guest still refuses to actually let me use the device, despite the device showing up in device manager after binding, and reporting that it's installed correctly. I'm hooking up a MIDI controller, it shows up with the right name under the right section, but every single application with MIDI controller support goes "there are no MIDI controllers connected".

zeerayne, 2023-01-06:
Maybe it is related to Hyper-V connection to guest? By default enhanced session mode (it is like RDP session) is used, and (maybe) MIDI controller work only in local session, but not enhanced session mode. Try to turn off `Allow enhanced session mode` is Hyper-V settings.

Pomax, 2023-01-06:
That was indeed the problem. With enhanced session mode turned off, things actually work. Hilariously, an ASIO device over USB then sounds _truly dire_ so I guess hyper-v is trying very hard to be completely useless for a nicely siloed audio VM