cezanne/usbip-win #333: Connection issue 22.08 and Win10

Mikilu80, 2022-12-26:
Goodday, Merry christmas!!

Following the youtube tutorail (https://www.youtube.com/watch?v=lsMsl8LEZvs&t=199s)
Im stuck at 8:40, I cant get connection with the host.

I found some suggestions about running usbipd, but i do not see this command being run in the video

And when trying to run "sudo usbipd -4 -d" as stated in the guide, i get the message:
"WARNING: usbipd not found for kernel 5.10.134"

I have tried to install the linux tools but failed so far...
Is 22.08 supported at all or should I not even try ?

I use this version of ubuntu on a Radxa Zero , so not all OS are available
https://wiki.radxa.com/Zero/downloads

Thx in advance.

PavelTLT, 2022-12-28:
Hello!
I have some stange problems with usbipd & ubuntu 22.xx
Example, with Jacarta USB key.

But other devices working normally.
usbipd installed on this manual: https://habr.com/ru/company/selectel/blog/668590/

shortly, with sudo:
apt-get install linux-tools-`uname -r`

find /lib/modules/$(uname -r) -name '*.ko'| grep usbip

get this result:
/lib/modules/5.4.0-100-generic/kernel/drivers/usb/usbip/usbip-core.ko
/lib/modules/5.4.0-100-generic/kernel/drivers/usb/usbip/usbip-host.ko
/lib/modules/5.4.0-100-generic/kernel/drivers/usb/usbip/vhci-hcd.ko
/lib/modules/5.4.0-100-generic/kernel/drivers/usb/usbip/usbip-vudc.ko

modprobe usbip-core
modprobe usbip-host
modprobe vhci-hcd

usbipd -D

usbip list -l

Mikilu80, 2022-12-29:
also tested with Raspberry Pi Zero and FIRST got same results

**On Pi Zero**
pi@raspberrypi:~ $ sudo usbip bind -b 1-1
usbip: info: bind device on busid 1-1: complete

**On Win10**
Ping statistics for 192.168.178.23:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 4ms, Maximum = 15ms, Average = 9ms
PS C:\Users\Pietje\Documents\usbip-win-0.3.6-dev> ./usbip.exe list -r 192.168.178.23
usbip: error: failed to connect a remote host: 192.168.178.23

Tried this from [https://github.com/cezanne/usbip-win/issues/185](url)
PS C:\Users\Pietje\Documents\usbip-win-0.3.6-dev> ./usbip.exe attach_ude -r -b 1-1
usbip: error: invalid command: attach_ude

On a earlier (closed) post I saw a log where indeed usbipd was called and succesfully run, so also did run it with scces this time. after I could see the list of remote usb devices.

Conclusion: usbipd is very much needed and not available in latest kernels ar linux-tools.

Pi zero is running on 5.10.17+, and i am not going to update.

closed 2022-12-29 by Mikilu80

ptorrent, 2023-10-03:
Did oyu find the solution ?

Mikilu80, 2023-11-10:
Hi,

yes got it working at the end, only problem is that at high data it fails,
For example when using laser engraver, i can only do cutting and not
engraving as the last one requires much more data to be send.

I wrote down these notes, hope it helps you.
"
LINUX HOST

login

sudo apt-get remove --purge usbip* libusbip*
sudo apt-get install linux-tools-generic
sudo apt-get install usbip
sudo modprobe vhci-hcd
sudo modprobe usbip-core
sudo modprobe usbip-host
sudo usbip list -l
sudo usbip bind -b 1-1

WINDOWS CLIENT
[DOWNLOAD usbip-win]

cd "C:\Users\Maarten\OneDrive - HSH\Documents\usbip-win-0.3.6-dev"
./usbip.exe list -r 192.168.178.23
./usbip attach -r 192.168.178.23 -b 1-1
"

I now made a batch file that runs them on startup.

Good luck