cezanne/usbip-win #345: install usbip vhci driver cause a BSOD

LinYan94, 2023-03-17:
When I install the usbip vhci dirver, it cause my computer BSOD. What can I do to resolve this problem?
`IRP_MN_QUERY_LEGACY_BUS_INFORMATION`
This is the windbg print and log print. 
`usbip_vhci: DriverEntry: Enter
usbip_vhci: RegistryPath FFFFA10118875000
usbip_vhci: DriverEntry: Leave
usbip_vhci: adding root: pdo: 0xFFFFA1011396CCF0
usbip_vhci: creating child cpdo
usbip_vhci: root added: vdev: FFFFA1010EE60EE0
usbip_vhci: vhci_pnp(root): Enter: minor:IRP_MN_QUERY_LEGACY_BUS_INFORMATION, irp: FFFFA10115546E50
************************************************************
Driver Verifier detected violation:

Caller has completed untouched IRP_MJ_PNP (instead of passing the irp
down) or non-PDO has failed the irp using incorrect value of
STATUS_NOT_SUPPORTED.

CulpritAddress = FFFFF806E5968960, Irp = FFFFA10115546E50.
************************************************************

Break instruction exception - code 80000003 (first chance)
nt!ViErrorFinishReport+0x8e:
fffff807`5afda2ca cc              int     3`
The stack is here.
[image]

LinYan94, 2023-03-20:
I use wdm driver in this problem.

vadimgrn, 2023-03-26:
Pay attention to [this](https://github.com/vadimgrn/usbip-win2) project, please.

milomai, 2024-06-25:
This ocurre when I install driver in Device Manager manully.
But it's OK by run `usbip.exe install`