cezanne/usbip-win #346: how to use  usb device that is redirected from server to client

elvis-sun, 2023-03-27:
一.  Server
1. usbip.exe bind -b {busid}   [OK]
2. usbipd.exe -d -4                 [OK]
二. Client
  usbip.exe attach -r {ip} -b {busid}    [OK]
  succesfully attached to port 0

I see the device [USB/IP Stub] under system devices in device manager
My question is that:
      How to use the redirected usb device？

elvis-sun, 2023-03-27:
> How to use the redirected usb device？

My device is usb storage disk

vadimgrn, 2023-03-27:
New drive letter should appear in the system (D:, E:, ...). Use it like any other drive.

elvis-sun, 2023-03-28:
> New drive letter should appear in the system (D:, E:, ...). Use it like any other drive.

@vadimgrn
thanks for your reply.
I have one USB printer, that is a usb composite device(printer and scanner). 
Just recognize it as scanner, not include printer, after redirected.

vadimgrn, 2023-03-28:
Try UDE or WDM driver from [this](https://github.com/vadimgrn/usbip-win2) project, pls.

elvis-sun, 2023-03-30:
> Try UDE or WDM driver from [this](https://github.com/vadimgrn/usbip-win2) project, pls.

Thank you