cezanne/usbip-win #350: Attaching remote device by VID rather than BUS ID

fragtion, 2023-07-27:
As per title, it would be nice to have the option built in to have the ability to select a remote device (to attach etc) by its VID, not only the BUS ID. This can help in setups where resilience is needed for a single USB device potentially shifting USB port

usbipd supports binding a device in this way, so it seems to make a lot of sense to allow this on the client side as well for a seamless/consistent setup on both ends

vadimgrn, 2023-07-30:
Linux 'usbip bind' uses --busid only, as well as usbip attach'. pid/vid is not unique.

vadimgrn, 2023-07-31:
See also https://github.com/torvalds/linux/blob/master/tools/usb/usbip/src/usbip_network.h#L62,

fragtion, 2023-07-31:
@vadimgrn  Thanks for taking the time to provide such an on-point response. That is interesting indeed.

In response I will say that it is interesting how usbipd allows binding by VID if, in theory, it's possible for a "non-unique" exception to be raised on the host side as well.  I still find this inconsistency between the client and host to be strange, and I still maintain that it would be a convenient feature to allow binding by VID on the client side. If there are more than 1 device with the same VID, the client could just throw an error. But if there's one device that matches, I don't see why the method can't be used.

But yes, ideally the patch would then be applied upstream at the linux codebase first... So I guess it's up to this project's maintainers to decide if an exception should be made for the windows version or not.

Cheers & thanks again.

vadimgrn, 2023-07-31:
I do not understand where did you get info about support for bind or attach by VID?
Kernel 6.2.0-26-generic

```~$ usbip bind -h
usage: usbip bind <args>
    -b, --busid=<busid>    Bind usbip-host.ko to device on <busid>
vadym@pc:~$ usbip attach -h
usage: usbip attach <args>
    -r, --remote=<host>      The machine with exported USB devices
    -b, --busid=<busid>    Busid of the device on <host>
    -d, --device=<devid>    Id of the virtual UDC on <host>
```

fragtion, 2023-07-31:
> I do not understand where did you get info about support for bind or attach by VID? Kernel 6.2.0-26-generic

`C:\WINDOWS\system32>usbipd bind --help
usbipd-win 2.4.1

Description:
  Registers a single USB device for sharing, so it can be attached to other machines. Unless the --force option is
  used, shared devices remain available to the host until they are attached to another machine.

  Exactly one of the options '--busid' or '--hardware-id' is required.

Usage:
  usbipd bind [options]

Options:
  -b, --busid <BUSID>          Share device having <BUSID>
  -f, --force                  Force binding; the host cannot use the device
  **-i, --hardware-id <VID:PID>  Share device having <VID>:<PID>**
  -?, -h, --help               Show help and usage information
`
https://github.com/dorssel/usbipd-win

vadimgrn, 2023-07-31:
This is Windows implementation of usbipd which is not canonical. In my [project](https://github.com/vadimgrn/usbip-win2) I'm considering Linux compatibility only. This project is abandoned defacto.