cezanne/usbip-win #356: Xbox Wireless Adapter for Windows Attaches but unable to associate controller

jshank, 2023-11-17:
**Server Version:** usbip-win-0.3.6-dev
**Server OS Version**: Win10 22H2 19045.3693 
**Server Driver Version:** usbip-win VHCI(ude) Driver Version 0.3.6.0
**Client Version:** usbip (usbip-utils 2.0) from usbip/stable,now 2.0+6.1.52-1+rpi1 armhf
**Client OS Version:** Linux usbip 6.1.0-rpi4-rpi-v7 #1 SMP Raspbian 1:6.1.54-1+rpt2 (2023-10-05) armv7l GNU/Linux (RPi 3B+)

Attaching the port appears successful, showing the Xbox Wireless Adapter for Windows correctly. However, I'm unable to pair an Xbox Wireless Controller with the adapter. The previously paired controller flashes and there is no more activity in the logs (the adapter does not light up when pairing button pressed). I've confirmed a wired mouse works correctly so I know the stack works. Interestingly, the XBox adapter gets insanely hot after about 20-min.

I have full traceview logs but nothing interesting happens in 44,000 lines of trace. Let me know if I should attach them. Any ideas? Let me know if I can include any other diag output that may help get to the root of this. 

**Xbox Wireless adapter for Windows Properties - Events**
```
Device USB\VID_045E&PID_02FE\206105 was started.

Driver Name: oem73.inf
Class Guid: {4d36e972-e325-11ce-bfc1-08002be10318}
Service: mt7612US
Lower Filters: 
Upper Filters: xboxgip
```

**usbip list -l**
```
usbip list -l
 - busid 1-1.1 (0424:ec00)
   Microchip Technology, Inc. (formerly SMSC) : SMSC9512/9514 Fast Ethernet Adapter (0424:ec00)

 - busid 1-1.2 (045e:02fe)
   Microsoft Corp. : unknown product (045e:02fe)
```

**dmesg (client)**
```
[90333.222041] usb 1-1.2: new high-speed USB device number 14 using dwc_otg
[90333.353199] usb 1-1.2: New USB device found, idVendor=045e, idProduct=02fe, bcdDevice= 1.00
[90333.353234] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[90333.353252] usb 1-1.2: Product: XBOX ACC
[90333.353265] usb 1-1.2: Manufacturer: Microsoft Inc.
[90333.353278] usb 1-1.2: SerialNumber: XXXXXXX
[90333.452012] usb 1-1.2: reset high-speed USB device number 14 using dwc_otg
[90333.583370] mt76x2u 1-1.2:1.0: ASIC revision: 76320044
[90333.612344] mt76x2u 1-1.2:1.0: firmware: direct-loading firmware mt7662_rom_patch.bin
[90333.612379] mt76x2u 1-1.2:1.0: ROM patch build: 20141115060606a
[90333.775497] mt76x2u 1-1.2:1.0: firmware: direct-loading firmware mt7662.bin
[90333.775530] mt76x2u 1-1.2:1.0: Firmware Version: 0.0.00
[90333.775548] mt76x2u 1-1.2:1.0: Build: 1
[90333.775561] mt76x2u 1-1.2:1.0: Build Time: 201507311614____
[90333.958204] mt76x2u 1-1.2:1.0: error: MCU resp evt:9 seq:1-0
[90333.958295] mt76x2u 1-1.2:1.0: error: MCU resp evt:9 seq:1-0
[90333.958370] mt76x2u 1-1.2:1.0: error: MCU resp evt:9 seq:1-0
[90334.665852] ieee80211 phy7: Selected rate control algorithm 'minstrel_ht'
```
**usbipd output**
```
usbipd: info: connection from 10.1.10.15:61404
usbipd: info: received request: 0x8003(5)
usbipd: info: found requested device: 1-1.2
usbip: info: connect: 1-1.2
usbipd: info: request 0x8003(5): complete
```

jshank, 2023-11-17:
I was able to get this working with VirtualHere without issue so far but I'd much rather solve for OSS projects like this one. Please let me know if there are any captures or additional data I can provide.

vadimgrn, 2023-11-20:
Try [this](https://github.com/vadimgrn/usbip-win2/tree/master) project, please.

aazzolini, 2023-12-02:
I am running into the same issue. @vadimgrn , I also tried usbip-win2 but in that case it doesn't even show the device on device manager. (Also, I couldn't get the mouse to move using usbip-win2, although it showed up as a device).

awdAvenger, 2023-12-31:
I just want to add that I got it working with usbip-win2, but since it's not signed it's not really a feasible solution as that stops other software from working correctly.

I did block the mt76x2u module on the host though. Not sure if that was required or not.

vadimgrn, 2023-12-31:
> but since it's not signed it's not really a feasible solution as that stops other software from working correctly.

It does not stop other software from working correctly (see [here](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/the-testsigning-boot-configuration-option#behavior-of-windows-when-loading-test-signed-code-is-enabled)), but it increases the vulnerability of a system.

awdAvenger, 2023-12-31:
> > but since it's not signed it's not really a feasible solution as that stops other software from working correctly.
> 
> It does not stop other software from working correctly (see [here](https://learn.microsoft.com/en-us/windows-hardware/drivers/install/the-testsigning-boot-configuration-option#behavior-of-windows-when-loading-test-signed-code-is-enabled)), but it increases the vulnerability of a system. Most company policies prohibit enabling TESTSIGNING on computers. Private users decide for themselves whether to enable test signing or not.

Many anti-cheat solutions won't work when test signing is enabled, which hinders the use case for this device somewhat.

I know it's not straight forward to get it signed though.

thrynae, 2024-02-25:
In my case the wireless adapter shows up in device manager without issue, but I can't connect a controller either (with either this repo, or the usbip-win2 repo). At this point I would even consider trying a WSL pass-through, although I haven't seen any mentions of success with that route.

@vadimgrn did you test an Xbox wireless adapter? What steps should I be taking to make it work? Should I switch to the WDM driver?

vadimgrn, 2024-02-25:
I don't have Xbox for the testing. Without the driver's log I can't figure out what is wrong.

thrynae, 2024-02-28:
I installed VS 2022, the SDK, and the WDK, after which I ran the batch script in your readme, but I get the messages below (even after I create the usbip folder in TEMP (to hide my username I put back %TEMP%)):

> C:\Windows\System32>tracelog.exe -stop usbip-flt
> Operation Status:       4201L   The instance name passed was not recognized as valid by a WMI data provider.
> Logger Name:            usbip-flt
> C:\Windows\System32>tracelog.exe -stop usbip-ude
> Operation Status:       4201L   The instance name passed was not recognized as valid by a WMI data provider.
> Logger Name:            usbip-ude
> C:\Windows\System32>tracefmt.exe -nosummary -p %TEMP%\usbip -o usbip-flt.txt usbip-flt.etl
> Setting log file to: C:\Windows\System32\usbip-flt.etl
> Cannot open log file for reading
> C:\Windows\System32>tracefmt.exe -nosummary -p %TEMP%\usbip -o usbip-ude.txt usbip-ude.etl
> Setting log file to: C:\Windows\System32\usbip-ude.etl
> Cannot open log file for reading

How should I get the log to you? (and would you like me to open an issue on your own repo?)

Alternatively, how can I generate the log for this repo? Just in case the cause is the same and the issue can be fixed for both at once.

vadimgrn, 2024-02-28:
If you are using (usbip-win)[https://github.com/cezanne/usbip-win], I can't help you, this project is dead. I'm developing (usbip-win2)[https://github.com/vadimgrn/usbip-win2/tree/master?tab=readme-ov-file] driver.

thrynae, 2024-02-28:
The log I posted was for the usbip-win2 driver (since that repo mentions a batch script to capture the log, while the usbip-win repo does not).