cezanne/usbip-win #364: Camera issues

goodusername5, 2024-04-09:
Does anyone know why streaming from attached webcams fail so often? BSOD, driver failures etc.
Any work arounds out there?

vadimgrn, 2024-04-09:
Use this https://github.com/vadimgrn/usbip-win2/tree/develop

goodusername5, 2024-04-10:
Hey @vadimgrn, I've tried using yours but I get the same BSOD crash on webcam stream. I've run !analyze in WinDbg:
`DPC_TIMEOUT_TYPE:  DPC_QUEUE_EXECUTION_TIMEOUT_EXCEEDED
FAULTING_SOURCE_LINE:  D:\usbip-win2\drivers\ude\device_queue.cpp
FAULTING_SOURCE_FILE:  D:\usbip-win2\drivers\ude\device_queue.cpp
FAULTING_SOURCE_LINE_NUMBER:  166
FAULTING_SOURCE_CODE:  
No source found for 'D:\usbip-win2\drivers\ude\device_queue.cpp'
SYMBOL_NAME:  usbip2_ude!usbip::device::remove_egress_request+34
MODULE_NAME: usbip2_ude
IMAGE_NAME:  usbip2_ude.sys
nt!KeBugCheckEx
nt!KeAccumulateTicks+0x239
nt!KiUpdateRunTime+0xd1
nt!KeClockInterruptNotify+0xc1
nt!HalpTimerClockInterrupt+0x10a
nt!KiCallInterruptServiceRoutine+0x9c
nt!KiInterruptSubDispatchNoLockNoEtw+0xfa
nt!KiInterruptDispatchNoLockNoEtw+0x37
nt!KxWaitForSpinLockAndAcquire+0x1c
nt!KeAcquireSpinLockRaiseToDpc+0x8e
Wdf01000!imp_WdfSpinLockAcquire+0x70 [minkernel\wdf\framework\shared\support\fxspinlockapi.cpp @ 142] 
usbip2_ude!usbip::device::remove_egress_request+0x34 [D:\usbip-win2\drivers\ude\device_queue.cpp @ 166] 
usbip2_ude!`anonymous namespace'::endpoint_purge+0x10c [D:\usbip-win2\drivers\ude\device.cpp @ 156] 
udecx!Endpoint_QueuePurgeOrAbortComplete+0xa3
Wdf01000!FxIoQueueIoState::Invoke+0x4e [minkernel\wdf\framework\shared\inc\private\common\FxIoQueueCallbacks.hpp @ 126] 
Wdf01000!FxIoQueue::ProcessIdleComplete+0x9e [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 4739] 
Wdf01000!FxIoQueue::DispatchEvents+0xf03b [minkernel\wdf\framework\shared\irphandlers\io\fxioqueue.cpp @ 2920] 
Wdf01000!imp_WdfRequestForwardToIoQueue+0xed54 [minkernel\wdf\framework\shared\core\fxrequestapi.cpp @ 3130] 
usbip2_ude!usbip::device::move_egress_request_to_queue+0x66 [D:\usbip-win2\drivers\ude\device_queue.cpp @ 176] 
usbip2_ude!`anonymous namespace'::move_request_to_queue+0x23 [D:\usbip-win2\drivers\ude\device_ioctl.cpp @ 40] 
usbip2_ude!`anonymous namespace'::send_complete+0x94 [D:\usbip-win2\drivers\ude\device_ioctl.cpp @ 82] 
nt!IovpLocalCompletionRoutine+0x16e
nt!IopfCompleteRequest+0x134
nt!IovCompleteRequest+0x1e1
nt!IofCompleteRequest+0x1e6d39
afd!WskProTLSendOrDisconnectComplete+0x6f
tcpip!TcpTcbReceive+0x6c0
tcpip!TcpMatchReceive+0x519
tcpip!TcpReceive+0x35f
tcpip!TcpNlClientReceiveDatagrams+0x16
tcpip!IppProcessDeliverList+0xb6
tcpip!IppReceiveHeaderBatch+0x3a9
tcpip!IppReceivePackets+0x502
tcpip!FlpReceiveNonPreValidatedNetBufferListChain+0x2f1
tcpip!FlReceiveNetBufferListChainCalloutRoutine+0x198
nt!KeExpandKernelStackAndCalloutInternal+0x7a
nt!KeExpandKernelStackAndCalloutEx+0x1d
tcpip!FlReceiveNetBufferListChain+0x52b
ndis!ndisMIndicateNetBufferListsToOpen+0x141
ndis!ndisDoPeriodicReceivesIndication+0x3ae
ndis!ndisPeriodicReceivesWorker+0x73
ndis!ndisReceiveWorkerThread+0x86a6
nt!PspSystemThreadStartup+0x57
nt!KiStartSystemThread+0x34`

Any idea what's causing this?

vadimgrn, 2024-04-10:
Please add full output from the debugger as described [here](https://github.com/vadimgrn/usbip-win2?tab=readme-ov-file#debugging-bsod)