cezanne/usbip-win #369: connected device is not realy connected (W11 / 24H2)

sugramm, 2024-06-23:
I have set up the USBIP server on a Raspberry Pi 4.
I have finally managed to connect the device in Windows
For testing I wanted to connect a USB stick

[image]

but I can't see / recognize the device anywhere under Windows

[image]

When I then check whether the device is there, the list is empty?!

[image]

Can someone tell me where the error lies here?

NullYing, 2024-07-25:
usb3.0 or usb2.0?maybe you can use usb2.0 and try agent

it-parasite, 2024-11-12:
Hi!
Is the situation the  with USB sticks from other manufacturers? Is this problem only in PowerShell or in CMD too? What USBIP drivers (release number, UDE/WDM) and how did you install them?