cezanne/usbip-win #372: Build Issue in Windows

chandraOloid, 2024-09-12:
Getting this error when trying to build *usbip-win* with visual Studio 2022 10.0.18.xx SDK&/WDK. 

Tried everything but not ablet to solve this issue. 

Projects *usbip_vhci* and *usbip_vhchi_ude* have this error. 

```
Severity	Code	Description	Project	File	Line	Suppression State	Details
Error	MSB3073	The command "gencat.bat C:\usbip-win\Release\x64\ 10_x64 C:\usbip-win\/driver/usbip_test.pfx

:VCEnd" exited with code -1.	usbip_vhci	C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets	166		
```

Kindly help ASAP.

ddbb2017, 2025-01-24:
You need to specify the exact location for the `gencat.bat`. Search for it in all the folders. It is not in the root for the solution. These files will be different from each driver project.

chandraOloid, 2025-01-28:
Solved. Just needed to look deeper. Again many thanks @ddbb2017

closed 2025-01-28 by chandraOloid